chrisstaite/TeensyDmx #1: Pin details for example required

peternewman, 2016-12-09:
Hi,

I tried to get this working a little while ago, but couldn't get DMX Tx or Rx working, or RDM comms.

Can you confirm which pins need connecting to what etc please?

peternewman, 2017-01-01:
I think this was due to https://github.com/chrisstaite/TeensyDmx/pull/2 as I've now got DMX Tx working.

chrisstaite, 2017-01-02:
Thanks for the patch, looks fine.  I've merged the pull request as you're probably aware.

closed 2017-01-02 by chrisstaite

peternewman, 2017-01-02:
Thanks @chrisstaite . I've got an example too which I'll open a PR for shortly.

harryprayiv, 2017-01-02:
Please, please, please, please do a example of Tx with an array of values.

chrisstaite, 2017-01-02:
Someone has produced one based on their version that works on all three serial ports.  They tell me that it will be added soon.

harryprayiv, 2017-01-02:
Yeah I saw that. Thanks to both of you. Much respect.

peternewman, 2017-01-02:
As promised: https://github.com/chrisstaite/TeensyDmx/pull/3

harryprayiv, 2017-01-02:
In my case, I have a TI SN75LBC184D wired to pins 0 for Rx, 1 for Tx (Hardware Serial1), and 24 for my REDE pin.

[image]

harryprayiv, 2017-01-03:
I have one question on the example you posted.  It seems from the fact that the first value is a in setChannels(); is a '0' that we are using array index notation to indicate the first channel.  **So, a '0' in that field would actually be setting the '1'channel and a '1' would be channel '2' and so on?**  

From my understanding, DMX channel numbers start at #001.  If I have to subtract one to keep with some under the hood requirement for array notation satisfied, I don't mind.  I would, however, recommend changing that in the library, someday , because it is less confusing to noobs like me.  If it has to be that way, it's no big deal.

Anyway, thanks for posting this example.  that Dmx.loop(); had escaped my notice.  

I will be posting some more advanced iterations of code using this library from time to time.

peternewman, 2017-01-03:
I was generally announcing it on here @harryprayiv as you'd asked; @chrisstaite should see the PR anyway I hope.

Yes, it's using zero based arrays, or alternatively you could think of it as the offset from the first slot (which can then be either slot 0 or slot 1 as you prefer). Wikipedia refers to slot 0 containing the start code (but then also call it the first slot):
https://en.wikipedia.org/wiki/DMX512#Protocol
The DMX512 standard seems to agree with this. I think the array passed in would always want to be zero indexed. I guess the first argument to setChannels could be renamed to offset, or some maths could be done internally to subtract one, but personally I think that's just as likely to cause more issues and probably confuse more people.

The loop method doesn't actually (currently) do anything during transmission, only reception, but I put it in while troubleshooting, and it probably makes it less confusing for people in future to leave it in there. I think it increases the chances of people getting a working DMX receive sketch if they modify the existing one.

chrisstaite, 2017-01-03:
It is 0 based for two reasons:

1) It would require subtraction by one in the library which may be inefficient if the user is using 0-based as well and hence will be adding by one
2) The library is compatible with the Atmel library which operates on 8-bit processors and 511 fits in 8 bits and 512 requires 9 bits, therefore requiring the value to be passed as two bytes rather than 1 and this is unlikely to be changed in the base library that this is a port of.

chrisstaite, 2017-01-03:
I've just added a bit more code to make RDM actually functional and another example.  I've also removed the requirement for the RDM structure for transmit only.

harryprayiv, 2017-01-03:
I just want to sincerely thank both of you for all of your changes the past two days.  I will have a DMX send testing setup ready tomorrow, so I will be able to let you know what works and what doesn't.  However, I'm pretty sure everything will work fine.  

_Just so both of you are aware, I will be testing this library with Teensy 3.6 on my custom designed board._  

I am starting to compile a list of commands (with example usage) for the people like me.  **I really appreciate the addition of setDMXchannel(); and setDMXchannels()**; and the changes to the requirements of that RDM struct.