chrisstaite/TeensyDmx #4: Trouble with changing values in the loop

harryprayiv, 2017-01-07:
I finally have a working test setup with this library and it **definitely** works.  

However, it doesn't update whenever the arduino loops.  

When the Teensy goes around the loop, the array values don't change no matter how much I try in my code.  I am thinking that the array that temporarily holds the DMX values is either a constant instead of a variable or doesn't actually transfer new values into the buffer once it has them.  I am not that great at this stuff so I will just post the example that I have been working on.

Another thing is the DMX REDE pin. I am not sure if I have it hooked up right.  Is it the Read/Write enable switch for use with RDM so you can do both send and recieve?  If so, I have tried both pin 2 and pin 24 and the code acts EXACTLy the same.  I THINK I have REDE on 24, but I am not sure since it doesn't do anything anyway.  

Here's my simplified code in an attempt to get at the heart of this matter more efficiently.

```
#include <TeensyDmx.h>

#define DMX_REDE 24

TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(8);
  Dmx.setMode(TeensyDmx::DMX_OUT);

}

void loop() {
  uint8_t dmxVal[] = {analogRead(1)}; // this acts the same if you use bytes
  
  Dmx.setChannels(0,dmxVal,1);
  Dmx.loop(); // acts the same with or without this command
  //should I be somehow releasing the current DMX data here and refreshing it when it comes back around in the loop?  IF so, how...(and WHY  actually)?
}
```

harryprayiv, 2017-01-07:
Here's another bit of code that does the same thing (_my slide pots are mounted upside down so I use that mapping and value crunching to eat up some clock cycles in order to give the analog pins time to settle to 0_).

I am populating an array and (I think, after the loop-through re-reading and repopulating those arrays with new values.  Correct me if I am wrong here.

`TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(8);
  Dmx.setMode(TeensyDmx::DMX_OUT);

}

void loop() {

//_____________________________________________________________________________// DMX 8 bit Array Code (begin)
  rawVal[0] = {analogRead(1)};
  displayVal[0] = map(rawVal[0],1,256,255,1);
  dmxVal[0] = displayVal[0];
//  Dmx.setDmxChannel(1,dmxVal[0]);  //instead?
  
  
  rawVal[1] = {analogRead(2)};
  displayVal[1] = map(rawVal[1],1,256,255,1);
  dmxVal[1] = displayVal[1];
//  Dmx.setDmxChannel(2,dmxVal[1]);  //instead?

  rawVal[2] = {analogRead(3)};
  displayVal[2] = map(rawVal[2],1,256,255,1);
  dmxVal[2] = displayVal[2];
//  Dmx.setDmxChannel(3,dmxVal[2]);  //instead?
  
  
  rawVal[3] = {analogRead(4)};
  displayVal[3] = map(rawVal[3],1,256,255,1);
  dmxVal[3] = displayVal[3];
//  Dmx.setDmxChannel(4,dmxVal[3]);  //instead?

  rawVal[4] = {analogRead(5)};
  displayVal[4] = map(rawVal[4],1,256,255,1);
  dmxVal[4] = displayVal[4];
//  Dmx.setDmxChannel(5,dmxVal[4]);  //instead?
  
  rawVal[5] = {analogRead(6)};
  displayVal[5] = map(rawVal[5],1,256,255,1);
  dmxVal[5] = displayVal[5];
//  Dmx.setDmxChannel(6,dmxVal[5]);  //instead?

  rawVal[6] = {analogRead(7)};
  displayVal[6] = map(rawVal[6],1,256,255,1);
  dmxVal[6] = displayVal[6];
//  Dmx.setDmxChannel(7,dmxVal[6]);  //instead?
    
  rawVal[7] = {analogRead(8)};
  displayVal[7] = map(rawVal[7],1,256,255,1);
  dmxVal[7] = displayVal[7];
//  Dmx.setDmxChannel(8,dmxVal[7]);  //instead?

  rawVal[8] = {analogRead(9)};
  displayVal[8] = map(rawVal[8],1,256,255,1);
  dmxVal[8] = displayVal[8];
//  Dmx.setDmxChannel(9,dmxVal[8]);  //instead?
  
    Dmx.setDmxChannels(1,dmxVal,9);
}`  

Anyway, They are all behaving the same way.  They do what I want.  I just have to reboot the Teensy to get them to refresh their Dmx.setDmxChannel(), Dmx.setChannel(), Dmx.setDmxChannels(), and Dmx.setChannels()  commands.  I even tried sending a 0 value on all channels after a long delay but it still didn't do anything.

harryprayiv, 2017-01-07:
I am working on a sort of "blink without delay" state machine right now to make sure that I can allow your library to have adequate time to transmit or receive.  Which leads me to this question: how long does a single 1-512 instruction take at 250,000 baud?  Can we make a blink without delay type of algorithm that can wait for a full "frame" to be transmitted?

pppaule, 2017-01-07:
Hey Harry,

You have to grab the stuff out of the array, what makes IMHO the setChannels impractical for arrays. Something like this would do the job:

```
//_____________________________________________________________________________// DMX 8 bit Array Code (begin)

for (int i = 0; i < 9; i++) {
  rawVal[i] = {analogRead(i+1)};
  displayVal[i] = map(rawVal[i],1,256,255,1);
  dmxVal[i] = displayVal[i];
  Dmx.setDmxChannel(i+1,dmxVal[i]);
}
```

Have fun, 
Daniel

harryprayiv, 2017-01-07:
I used that code but it somehow still doesn't alleviate the greater issue, which is that the code doesn't read the analog values continuously into the DmxValues.  I believe the dmx values are constants and they are locked into a private function.  I suppose I could try tinkering with the code in the .cpp file (I would learn a ton) but I would also break a ton of stuff, I imagine.    Here's the updated code from your snippet you sent.  It still behaves the same as the one I posted but is a hell of a lot more efficient.

```
#include <rdm.h>
#include <TeensyDmx.h>

#define DMX_REDE 24

byte dmxVal[512];
long rawVal[9];
long displayVal[9];

TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(8);
  Dmx.setMode(TeensyDmx::DMX_OUT);
}

void loop() {

  for (int i = 0; i < 9; i++) {
  rawVal[i] = {analogRead(i+1)};
  displayVal[i] = map(rawVal[i],1,256,255,1);
  dmxVal[i] = displayVal[i];
  Dmx.setDmxChannel(i+1,dmxVal[i]);
  }
}
```

chrisstaite, 2017-01-08:
`#include <TeensyDmx.h>

#define DMX_REDE 24

byte displayVal[8];

TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(8);
  Dmx.setMode(TeensyDmx::DMX_OUT);
}

void loop() {
  for (int i = 0; i < 9; ++i) {
    displayVal[i] = map(analogRead(i+1), 1, 256, 255, 1);
    Dmx.setChannel(i, displayVal[i]);
    Serial.print("Set channel ");
    Serial.print(i + 1, DEC);
    Serial.print(" to ");
    Serial.println(displayVal[i], DEC);
  }
}`

Does this code work on your hardware?

Are you getting a DMX signal through on your fixture (even if it's all zero)?

What if you set a channel in the setup() does that get sent?  If so, what if you set a channel in setup() and then change it to a different fixed value in the loop(), does it get the changed value from the loop?

harryprayiv, 2017-01-08:
Here's what happened with all of those tests you had me do: 

**setting levels in setup() but not in the loop**: worked for setting a single level
**setting levels in loop() but not setup(): works once:**  The value won't change if you make it a variable.
**setting levels in both loop and setup:** I only see the first value in the loop being set since it follows whatever my potentiometer was at at boot no matter what I hard set the levels to in setup

Here's the code that I used just now with both setup and loop (all serial readings are updating just fine)
```
#include <TeensyDmx.h>

#define DMX_REDE 24

byte displayVal[8];

TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(8);
  Dmx.setMode(TeensyDmx::DMX_OUT);
  Dmx.setChannel(0, 120);
}

void loop() {

////  for (int i = 0; i < 9; ++i) {
    displayVal[0] = map(analogRead(1), 1, 256, 255, 1);
    Dmx.setChannel(0, displayVal[0]);
    Serial.print("Set channel ");
    Serial.print(0 + 1, DEC);
    Serial.print(" to ");
    Serial.println(displayVal[0], DEC);
//  }
}
```

chrisstaite, 2017-01-08:
Been looking at the code, I think I found the bug...  Try updating to the latest version and see if that works...

harryprayiv, 2017-01-08:
Will do in a few. Thanks!

harryprayiv, 2017-01-08:
Your changes definitely are progress in the right direction.  There must be something wrong with that for loop because if I do the following, your code now works flawlessly: 

`#include<TeensyDmx.h>

#define DMX_REDE 24

byte displayVal[8];

TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(8);
  Dmx.setMode(TeensyDmx::DMX_OUT);
}

void loop() {
    displayVal[0] = map(analogRead(1), 1, 256, 255, 1);
    Dmx.setChannel(0, displayVal[0]);

    displayVal[1] = map(analogRead(2), 1, 256, 255, 1);
    Dmx.setChannel(1, displayVal[1]);

    displayVal[2] = map(analogRead(3), 1, 256, 255, 1);
    Dmx.setChannel(2, displayVal[2]);

    displayVal[3] = map(analogRead(4), 1, 256, 255, 1);
    Dmx.setChannel(3, displayVal[3]);

    displayVal[4] = map(analogRead(5), 1, 256, 255, 1);
    Dmx.setChannel(4, displayVal[4]);

    displayVal[5] = map(analogRead(6), 1, 256, 255, 1);
    Dmx.setChannel(5, displayVal[5]);    
}`

However, if I paste the code you sent me to test, I get nothing from the DMX.  I am thinking there must need to be a delay at the end of that loop or those analog pin numbers don't like to be assigned via that method...not sure what the issue is.

Just to be clear, here's the code that isn't working for some unknown reason
```
#include<TeensyDmx.h>

#define DMX_REDE 24

byte displayVal[8];

TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(8);
  Dmx.setMode(TeensyDmx::DMX_OUT);
}

void loop() {
  for (int i = 0; i < 9; ++i) {
    displayVal[i] = map(analogRead(i+1), 1, 256, 255, 1);
    Dmx.setChannel(i, displayVal[i]);
    Serial.print("Set channel ");
    Serial.print(i + 1, DEC);
    Serial.print(" to ");
    Serial.println(displayVal[i], DEC);
  }
}

```

harryprayiv, 2017-01-08:
It's so weird that if I introduce the for loop, the code stops works.  I am not sure why.  This code seems to function EXACTLY as needed, though.:

```
#include <TeensyDmx.h>

#define DMX_REDE 24

byte displayVal[8];
//int analogPin[9] = {1,2,3,4,5,6,7,8,9};

TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(16);
  Dmx.setMode(TeensyDmx::DMX_OUT);
}

void loop() {
    
    int i = 0;
    displayVal[i] = map(analogRead(1), 1, 65536, 255, 1);
    Dmx.setChannel(0, displayVal[i]);

    i = 1;
    displayVal[i] = map(analogRead(2), 1, 65536, 255, 1);
    Dmx.setChannel(1, displayVal[i]);

    i = 2;
    displayVal[i] = map(analogRead(3), 1, 65536, 255, 1);
    Dmx.setChannel(2, displayVal[i]);

    i = 3;
    displayVal[i] = map(analogRead(4), 1, 65536, 255, 1);
    Dmx.setChannel(3, displayVal[i]);

    i = 4;
    displayVal[i] = map(analogRead(5), 1, 65536, 255, 1);
    Dmx.setChannel(4, displayVal[i]);

    i = 5;
    displayVal[i] = map(analogRead(6), 1, 65536, 255, 1);
    Dmx.setChannel(5, displayVal[i]);

    i = 6;
    displayVal[i] = map(analogRead(7), 1, 65536, 255, 1);
    Dmx.setChannel(6, displayVal[i]);

    i = 7;
    displayVal[i] = map(analogRead(8), 1, 65536, 255, 1);
    Dmx.setChannel(7, displayVal[i]);

    i = 8;
    displayVal[0] = map(analogRead(9), 1, 65536, 255, 1);
    Dmx.setChannel(8, displayVal[i]);    
}
```

chrisstaite, 2017-01-08:
The third last line should have i rather than 0.

Could you try it without the loop, but with the analogRead(i + 1)?

harryprayiv, 2017-01-08:
ooof.  I think I was using the wrong code, screwing up my for loop.  Just a heads up: the code you asked me to try doesn't work because the for loop exceeds the size of the array by 1.

The following code works perfectly.  

`#include<TeensyDmx.h>

#define DMX_REDE 24

byte displayVal[8];

TeensyDmx Dmx(Serial1, DMX_REDE);

void setup() {
  analogReadRes(8);
  Dmx.setMode(TeensyDmx::DMX_OUT);
}

void loop() {
  for (int i = 0; i < 8; ++i) {
    displayVal[i] = map(analogRead(i+1), 1, 256, 255, 1);
    Dmx.setChannel(i, displayVal[i]);
    Serial.print("Set channel ");
    Serial.print(i + 1, DEC);
    Serial.print(" to ");
    Serial.println(displayVal[i], DEC);
  }
}`

chrisstaite, 2017-01-08:
Yup, my bad...  Glad it's all working now.

closed 2017-01-08 by chrisstaite