chrisstaite/TeensyDmx #6: Need a way to not toggle a pin when using for only input or output

ssilverman, 2017-01-31:
The API assumes that we need to toggle a pin for selecting input or output mode. For systems either without free pins or set up to only transmit or receive, a pin doesn't need to be toggled. I'm playing with different ways to do this cleanly, but maybe you've already thought of a good way.

ssilverman, 2017-01-31:
I was playing with making a class that can take a number as assignment, as a replacement for the uint_8 redePin parameter. Then that class would manage setting the pin state. Passing a nullptr for this could either cause the creating of a "null" implementation (in the spirit of the Null Pattern, would avoid having to check for nullptr for each call to the pin), or an actual nullptr that would be tested for each pin set.

Another, simpler, option is to change the type to int16_t and have negative values cause a nullptr to be assigned to m_redePin. Then, for each pin set, a check for nullptr would be necessary.

chrisstaite, 2017-01-31:
I've committed a solution which should work for you.

closed 2017-01-31 by chrisstaite