chrisstaite/TeensyDmx #7: m_uart.available() never returns a value > 0 for Teensy 3.1

ssilverman, 2017-02-01:
In short, the most basic receive example doesn't receive bytes. m_uart.available() never seems to return a value > 0.

The code:

```
#include "TeensyDmx.h"

#define CHANNEL 0
#define LED_PIN 13

TeensyDmx dmx(Serial1);

void setup() {
  dmx.setMode(TeensyDmx::DMX_IN);
  pinMode(LED_PIN, OUTPUT);
}

void loop() {
  dmx.loop();
  if (dmx.newFrame()) {
    Serial.println(dmx.getBuffer()[CHANNEL]);
  }
}
```

I've instrumented TeensyDmx.cpp to set `LED_PIN` to `HIGH` inside `readBytes()` and inside the while loop.

System setup:

1. Teensy 3.1.
2. Pin 0 (Serial1) is receiving DMX input.
3. Tested on an oscilloscope to make sure there's data.
4. Tested with another, much simpler, home-rolled library; it can receive data. Exact same hardware setup.

The "simpler" library only sets up an error handler and not the full status handler, as if `IRQ_UART0_ERROR` were defined in TeensyDmx.

ssilverman, 2017-02-01:
I neglected to add the DMX device I'm using: Enttec DMXUSB Pro.

chrisstaite, 2017-02-01:
I think I've figured out the issue.  It's not calling the ISR when it's not a frame error (i.e. for all of the actual packet).

Try the latest code and see if it works now.  I don't have a test rig currently so I can't give it a go myself.

ssilverman, 2017-02-01:
Now I see data, but it isn't reading what I expect. I'll report back soon...

ssilverman, 2017-02-01:
It looks like the `switch(m_state)` arrives at the default, "Discarding bytes" case.

ssilverman, 2017-02-01:
Exploring further, it never actually enters the `BREAK` state.

chrisstaite, 2017-02-01:
It should enter the BREAK state when UART0_S1 & UART_S1_FE is true.  Does that particular bit ever fire?

ssilverman, 2017-02-01:
Actually, yes it does, but it doesn't seem to enter the `case State::BREAK` section. Maybe this is intermittent... Could it be some sort of race condition when setting the states?

chrisstaite, 2017-02-01:
There shouldn't be a race condition as everything runs single-threaded.  However I wonder if there is an interrupt thing?  I realised that I wasn't enabling the framing error interrupt, so have done that now.  If it was firing, then that might not fix it, but it's worth a go...

chrisstaite, 2017-02-01:
It might help to write out the bytes that are read in readBytes() and a string when BREAK is set in completeFrame() and that should give an indication of how it's managed to go out of sync.

I imagine the problem is along the lines of it setting BREAK then reading the first byte which should be 0 to indicate DMX, but it getting out of sync and reading a value as the first byte instead which switches it into IDLE.

ssilverman, 2017-02-01:
Yeah, races with interrupts. Doing some more testing per your suggestion. I can't imagine Serial.print works when inside an ISR or when the interrupts are disabled, though. Glad there's an LED. :)

chrisstaite, 2017-02-01:
I've not tried, but it should be able to work as it just pushes the data onto a FIFO.

ssilverman, 2017-02-01:
This may not be related, but I also noticed that the very first time it gets a break signal, it counts that as having read one frame, so technically that's off by one.

chrisstaite, 2017-02-01:
Yeah, that frame count is just an indication.  It also overflows...

ssilverman, 2017-02-01:
I'm open to a voice chat if you like, so we can bounce ideas around more quickly.

ssilverman, 2017-02-01:
This feels like the right path upon which to look:

1. State starts at IDLE.
2. Break interrupt happens.
3. completeFrame() is called.
4. readBytes() is called within completeFrame().
5. Bytes are discarded because IDLE state.
6. State is set to BREAK at the end of completeFrame().

Meanwhile, loop() is called repetitively, thus, there's a chance, because the state is BREAK, that it gets reset to IDLE in readBytes(), preparing the system to discard more bytes the next time an actual break condition happens.

What do you think?

chrisstaite, 2017-02-01:
It's time for me to get some sleep. Open to work on it tomorrow evening or Friday.

ssilverman, 2017-02-01:
I'm not convinced that reading data from the uart is needed from the main program loop(). Why not fill the active buffer during the "data available" condition and not call TeensyDmx.loop()? Thinking aloud...

chrisstaite, 2017-02-04:
Thought about that, but best not to reply to RDM messages within an interrupt.  I'm not 100% sure if you even can...

Did you get any further?  I've tidied up the code a little so the Teensy-LC RX stuff isn't getting in the way of the Teensy 3.x RX.

chrisstaite, 2017-02-04:
if (UART0_S1 & UART_S1_FE)
    {
        (void) UART0_D;
        uart0_status_isr();

Been wondering if the problem lies in the above... I read from UART0 to clear the framing error, then call the status ISR to read the bytes.  I think I should be doing either one or the other...  Could you try both ways round (i.e. delete the (void) UART0_D, try it and then put it back and delete the uart0_status_isr).

harryprayiv, 2017-03-02:
I have a related issue it seems when updating to the latest library.  Good thing I saved a backup.  ;)

`Arduino: 1.8.0 (Mac OS X), TD: 1.34, Board: "Teensy 3.6, Serial, 180 MHz, Fast, US English"

PockonsoleV2WITHKeypad: In function 'void drawInterpolation()':
PockonsoleV2WITHKeypad:1015: warning: 'duration' is used uninitialized in this function 
   linear.setDuration(duration);
                               ^
PockonsoleV2WITHKeypad:1017: warning: 'currentTime' is used uninitialized in this function 
   drawpixel(currentTime, 62, false, false);
                                          ^
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp: In function 'void UART0RxStatus()':
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp:406:6: error: 'void TeensyDmx::completeFrame()' is private
 void TeensyDmx::completeFrame()
      ^
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp:993:41: error: within this context
         uartInstances[0]->completeFrame();
                                         ^
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp: In function 'void UART1RxStatus()':
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp:406:6: error: 'void TeensyDmx::completeFrame()' is private
 void TeensyDmx::completeFrame()
      ^
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp:1007:41: error: within this context
         uartInstances[1]->completeFrame();
                                         ^
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp: In function 'void UART2RxStatus()':
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp:406:6: error: 'void TeensyDmx::completeFrame()' is private
 void TeensyDmx::completeFrame()
      ^
/Users/harryprayiv/Documents/Arduino/libraries/TeensyDmx/TeensyDmx.cpp:1021:41: error: within this context
         uartInstances[2]->completeFrame();
                                         ^
Error compiling for board Teensy 3.6.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

chrisstaite, 2017-03-02:
Somehow I managed to miss those as friends.  It's fixed now...

harryprayiv, 2017-03-04:
Works on my end.  Compiles fine.  Thanks. I am also noticing about 1% less space used in my sketches that use this lib on Teensy 3.6 since this new change.  Looks like the code is getting even more efficient. 

When you get a chance, can you add some examples that illustrate the new functionality?

chrisstaite, 2017-03-04:
The two examples demonstrate all the public functions.  Have you got something else in mind?

closed 2017-03-04 by chrisstaite

harryprayiv, 2017-03-04:
No.  I thought maybe there was some change since OP said he/she was trying to do something unique in sending and not even attempting to receive.

I am currently writing a library (currently just a sketch filled with functions that I will need help turning in to a lib) that specifically deals with my hardware but I'd like to contribute to extending the functionality of TeensyDMX once I have something useful to you and generalists.  The best work I have done recently is using the Easing Library to interpolate between DMX values and animate them on my OLED screen.  I'd like to work on extending TeensyDMX to do Submasters (an array of ints or bytes for submasters and another array of booleans for channel selection?) and other common DMX functions that I am currently at work on.  Perhaps have a look at my Pockonsole library to see what I mean since I'm all over the place with features here.  ;)

chrisstaite, 2017-03-05:
It would be nice to see a library doing nice features, but for flexibility of platform it might be nice to make it work with any array of values, so it could be used with DMXSerial, TeensyDMX, FastLED, etc.