chrisstaite/TeensyDmx #8: Two Teensy 3.2 don't talk to each other

fricklr, 2017-03-09:
Hi,
I just tested your library on two Teensy 3.2 with Teensyduino 1.35. but sadly they can't talk to each other - the sender code from your example on it's own works, I tested it with a dimmer. 
Do you have any ideas?
Thank you very much

peternewman, 2017-08-25:
I think I decided the receive code is broken while trying to make an RDM Controller, I think I've sorted that bit (the RDM controller stuff still needs a fair bit more work). I'll see if I can pull that fix out as a patch and open a PR for it.

chrisstaite, 2017-08-26:
I've been meaning to look into this, but I've been very busy on other projects recently. Very happy to take pull requests.

peternewman, 2017-08-30:
Try this @fricklr , while @chrisstaite and I refine the PR if necessary before it gets merged https://github.com/chrisstaite/TeensyDmx/pull/12

peternewman, 2017-09-25:
Fixed via #14 @chrisstaite .

peternewman, 2017-09-27:
Care to close this one then @chrisstaite ?

chrisstaite, 2017-09-27:
If we could get someone else to test the new code, that would be good... @fricklr, are you able to test?

peternewman, 2017-09-27:
Fair point, admittedly I'd not actually tested a pair talking together...

fricklr, 2017-09-28:
Sorry, I'm without my gear for the next weeks. But it's  great news that it should work now. Thanks a bunch! 
Cheers

closed 2017-11-05 by chrisstaite