chrisstaite/TeensyDmx #22: Teensy LC not working

Thialf, 2017-10-26:
I've seen a Teensy LC transmit project, therefore I thought it could be possible to use a Teensy LC for receiving DMX, too.
It is compileable, but setting UART0_RWFIFO = 1; will fail immediatly, because Teensy LC does not have a Fifo. Uncommenting this line will lead to Dmx.setMode(TeensyDmx::DMX_IN); not hanging anymore.

But at receiving any data, it starts hanging once again, but I can't see why. Do you have an idea?

Teensy LC does not have IRQ_UART0_ERROR either. But as I can see, you already took care about it.

Thialf, 2017-10-28:
I've found several reasons. First, I needed Teensyduino >=1.28.
Second, the Teensy LC does only reset the frame error flag, if you write a 1 to it. 
Teensy LC does not have the 8 byte fifos at Serials. Therefore the __disable_irq(); call at readBytes() result in missing a lot of bytes and starts getting invalid telegrams. If this happens, the UART_SI_OR will be set and stops getting any more data. This flag has to be reset first.
Therefore I just resettet every errorflag at the end of the uart-rx-irq, regardless of there occurrence:
```
#ifndef IRQ_UART0_ERROR
void UART0RxStatus()
{
    ...
    UART0_S1 = UART_S1_IDLE | UART_S1_OR | UART_S1_NF | UART_S1_FE | UART_S1_PF;
}
```

m_uart.available(); is only boolean at Teensy LC. I had to use:
while (m_uart.available())
at readBytes()

chrisstaite, 2017-11-05:
I've re-designed the receive a little and taken on board your comments.  I've tested with two Teensy-LC's here and it seems to be working fine.  Could you confirm?

closed 2017-11-05 by chrisstaite

Thialf, 2017-11-13:
Yes, it's working here, too!