chrisstaite/TeensyDmx #31: Receiving DMX buffer shift?

maxdee, 2018-01-13:
Hello,
I am using this library to send and receive DMX. When receiving the array seems to be shifted right every frame. I am not sure what is causing this. 

Also thanks for making this! I am using this for work and would be happy to make a donation, please let me know how :)

Here is an example of the shifting, this pattern of 0s repeats, there should be no multiple zeros...

`0, 0, 0, 0, 0, 0, 0, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, `

`0, 0, 0, 0, 0, 0, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, `

`0, 0, 0, 0, 0, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, `

`0, 0, 0, 0, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, `

`0, 0, 0, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, `

`0, 0, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, `

`0, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, `

`172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203,`

chrisstaite, 2018-01-13:
Hi Max,

Could you send an example of the code you are using the library with.  There are two accessors for the DMX, one that is 0 indexed (like a standard C array) and one that is 1 indexed (like DMX channels).  I imagine the issue may be that you are confusing the two?

I've never put a donate link before, and I'm not opposed to it (I like beer money as much as the next person), however, with the contributions of Peter recently I'd feel bad taking all the credit.

Thanks, Chris.

maxdee, 2018-01-13:
Here is my receive code.
```
void receiveMode(){
    digitalWrite(STATUS_LED, millis()%1000 < 50);
    Dmx.loop();
    if (Dmx.newFrame()) {
        digitalWrite(LED, HIGH);
        int _index = -1;
        for(int i = 0; i < BUFFER_SIZE; i++){
            Serial.print(Dmx.getBuffer()[i]);
            Serial.print(", ");
        }
        Serial.println();
    }
    else {
        digitalWrite(LED, LOW);
    }
}
```

chrisstaite, 2018-01-13:
What are you using to transmit?  A desk?

maxdee, 2018-01-13:
Im actually using a pair of WDMX transmitters, each with a teensy. One is in transmit mode, receiving DMX and the other in receive mode, outputting DMX.

In prior test I did use a very cheap console to a optocoupled RS485 circuit and had a similar issue but very erratic.

Here is the transmit code I am using. 
```
void sendMode(){
    digitalWrite(STATUS_LED, millis()%200 < 50);
    counter++;
    delay(2);
    if(counter%20 == 1){
        digitalWrite(LED, HIGH);
        for(int i = 0; i < BUFFER_SIZE; i++){
            buffer[i] = ((millis()/20)+i)%255;
        }
        Dmx.setChannels(0, buffer, BUFFER_SIZE);
    }
    else {
        digitalWrite(LED, LOW);
    }
    Dmx.loop();
}
```

maxdee, 2018-01-14:
So I tested without the wireless dmx transceivers and get the same behavior. Do many other people use DMX receiving? Is there other code examples I could check out?
I can also run whatever tests needed.

chrisstaite, 2018-01-14:
I ran some basic tests and haven't found anything yet.  I'll hopefully have some time to do some more soon.

maxdee, 2018-01-14:
It seems to be related to the DMX break speed. In a setup without the WDMX transceivers, I can set `constexpr uint32_t BREAKSPEED = 100000;` to 10000 and it works. Does the breakspeed vary from hardware to hardware?

chrisstaite, 2018-01-14:
It shouldn't as it's divided by the CPU speed.  You've got your F_CPU set correctly I assume?

maxdee, 2018-01-14:
I do not know F_CPU, is it the frequency set by Tools->CPU Speed int the arduino menu?

chrisstaite, 2018-01-14:
Yes, it should be.

maxdee, 2018-01-14:
What should it be?

chrisstaite, 2018-01-14:
That depends on what the device is configured as.  Since you're using Teensy, it should be automatically correct.  Can you confirm which Teensy you are using?

maxdee, 2018-01-14:
3.2

maxdee, 2018-01-16:
FYI JimParis's DmxReceiver works.
https://github.com/jimparis/DmxReceiver

maxdee, 2018-01-16:
So far the only thing I suspect is that `void uart0_error_isr();` or `void UART0RxError(void)` do not get called for some reason. 

```
void uart0_error_isr();// Back reference to serial1.c
// UART0 will throw a frame error on the DMX break pulse.  That's our
// cue to switch buffers and reset the index to zero
void UART0RxError(void)
{
    // On break, uart0_status_isr() will probably have already
    // fired and read the data buffer, clearing the framing error.
    // If for some reason it hasn't, make sure we consume the 0x00
    // byte that was received.
    if (UART0_S1 & UART_S1_FE) {
        (void) UART0_D;
    }

    uartInstances[0]->completeFrame();
}
```

maxdee, 2018-01-17:
So if I print the data just after the buffer swapping in `completeFrame()` it has a consistent padding of 7 zeros.
```
0, 0, 0, 0, 0, 0, 0, 83, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 
0, 0, 0, 0, 0, 0, 0, 83, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 
0, 0, 0, 0, 0, 0, 0, 84, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 
```
If I print it in `getBuffer()` I get 
```
0, 0, 0, 0, 0, 0, 0, 230, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 
0, 0, 0, 0, 0, 0, 230, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 
0, 0, 0, 0, 0, 230, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 
0, 0, 0, 0, 231, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 
0, 0, 0, 231, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 
```

chrisstaite, 2018-01-17:
Thanks for your further research on this matter, I'll try to replicate at the weekend.

maxdee, 2018-01-17:
An other detail, seems to receive fine on a Teensy LC :)

chrisstaite, 2018-01-19:
I kind of expected that since I have two Teensy LC's as my test rig for the library :)

chrisstaite, 2018-01-19:
Assuming you are using Serial1 (i.e. UART0) could you try commenting out lines 1176-1180 which calls completeFrame.  On the Teensy-LC there isn't a separate interrupt for RxError, instead it calls RxStatus.  Therefore on the Teensy-LC we configure it to complete the frame in the status callback instead on in the error callback.

If that solves the issue then I imagine that the problem is my ifndef is not working.  I think that it might be worth standardising between the platforms and removing the RxError interrupt entirely if RxStatus is being called with the framing error interrupt.

You could additionally try removing the RxError interrupt call entirely and see if it still works for you (i.e. lines 1405 - 1413).

My issue is that I never have many Teensy's lying around to test with, they are all in projects.

maxdee, 2018-01-19:
Ok so commenting 1176-1180 leads to not receiving anything at all. But its a new lead, we will figure it out.

chrisstaite, 2018-01-20:
how about 1405 - 1413?

maxdee, 2018-01-20:
Commenting those results in receiving but still having offset issues.

chrisstaite, 2018-01-21:
Just to clarify the offset issue is happening on a Teensy 3.2, but the code works fine on a Teensy-LC?

I only question because IRQ_UART0_ERROR should be defined under the Teensy 3.2 build and given your comments above, it sounds like it isn't.

maxdee, 2018-01-21:
That is correct.

chrisstaite, 2018-01-21:
Could you try this patch?  I wonder if the framing error is coming in with the next byte?

```diff --git a/TeensyDmx.cpp b/TeensyDmx.cpp
index df6ab71..64b9ef8 100755
--- a/TeensyDmx.cpp
+++ b/TeensyDmx.cpp
@@ -1168,16 +1168,17 @@ void UART0RxStatus()
                }
        }
 #else
-       if (s & UART_S1_RDRF) {
-        uartInstances[0]->handleByte(UART0_D);
-    }
-#endif
 #ifndef IRQ_UART0_ERROR
-    if (s & UART_S1_FE)
-    {
+    if (s & UART_S1_FE) {
         (void) UART0_D;
         uartInstances[0]->completeFrame();
     }
+    else
+#else
+       if (s & UART_S1_RDRF) {
+        uartInstances[0]->handleByte(UART0_D);
+    }
+#endif
 #endif
     uart0_status_isr();
     // Reset all flags on Teensy-LC```

maxdee, 2018-01-21:
Can I pull it from anywhere?

maxdee, 2018-01-21:
Not sure this code works, or that I am making the correct modifications.

chrisstaite, 2018-01-21:
Pushed it to https://github.com/chrisstaite/TeensyDmx/tree/bugfix/31-rx-shift

maxdee, 2018-01-21:
Ok I tried it out and receive nothing, no new frame.

chrisstaite, 2018-01-21:
Looks like the error ISR isn't being triggered. I wonder if that define has been removed?

chrisstaite, 2018-01-21:
I found a bug, I was doing ifndef on an enum value, which obviously won't work.  I've made a change in the hope that fixes it...  Please try that.

maxdee, 2018-01-21:
That solves it! Tested on Teensy 3.2, TeensyDuino 1.4, Arduino 1.8.5

chrisstaite, 2018-01-21:
Great.  Glad we got to the bottom of that one.

closed 2018-01-21 by chrisstaite