chrisstaite/TeensyDmx #36: DMX TIMEOUT

frty130, 2019-06-12:
Disconnecting the DMX cable causes the led to freeze with the values of the last packet received. DMX serial and DMX Serial 2 have a timeout in the class. Can you add a timeout or point me in the direction of making one in my sketch using this class.

chrisstaite, 2019-07-04:
This is the purpose of the newFrame() function.  I don't want to add more complexity than required into the library (with RDM it's pretty huge already) and people might have different ways of implementing timers.  Therefore, I'd suggest calling this function and if it returns false then check the timeout hasn't passed.

chrisstaite, 2019-07-04:
As an exemplar here's a bit of (untested) code to do something like that:

```
void loop()
{
  static long lastFrame = millis();
  static constexpr long TIMEOUT = 5 * 1000;  // 5 seconds
  if (dmx.newFrame())
  {
    lastFrame = millis();
    // Do stuff with the frame
  }
  else if (lastFrame + TIMEOUT >= millis())
  {
    // Timeout reached, reset the LED or whatever
  }
}
```

closed 2019-07-04 by chrisstaite